Ralim/IronOS #2125: Pinecil V2 fails to negotiate PD or QC with a specific tip

c2h5oh, 2025-05-16:
**Describe the bug**
tl;dr; I have an original, lightly used Pinecil V2 tip that works perfectly fine with one exception: it breaks PD/QC leaving Pinecil stuck at 5V basic power.

I have noticed that my Pinecil V2 suddenly got stuck on 5V and would not negotiate anything else with a pine64-recommended power adapter (Amazon Basic 65W GaN) and USB-C cable I got from pine64 with the iron. I have tested 2 other known good adapters, one PD and one QC3 and with 2 other known good cables with exact same outcome: 5V and PD debug showing stage 3 and no other info.
With that I was getting low voltage warnings and indicated tip temperature would slowly climb to just under 110C, which was expected given the amount of power available to the iron.

I have checked all info available via IronOS menus and found nothing that was clearly wrong. I've reset to default settings - no change. I've upgraded to 2.22 and then to 2.34-rc4 in hopes that I will get better debug info / settings either on device or via bluetooth - no change. I've played around with new PD settings - no change.

At this point I was convinced Pinecil I was about to use for the third time since I've bought it was broken and I was just going through the motions for the sake of completeness before reaching out to seller: 
- I removed and reinstalled the tip - no change 
- Tried a a different tip - boom, all fixed
- Switched back to the original tip - same problem again

Huh, that tip is detected, looks like it's working, temp is shown, just power delivery gets broken. I would have never suspected the tip that has been used for a total of maybe 15 minutes to be the cause, nor I had any indication it could have been - and I think I should have received some kind of a warning.

**To Reproduce**
1. Have a weird, possibly half-broken tip that gets detected, reports it's heating up over 100C while remaining cool to the touch and causing anything beyond basic USB power to not work ;-)

**Expected behavior**
- a faulty tip is detected and reported as such, or
- voltage negotiation works

**Details of your device:**
 - Device: Pinecil V2
 - Release: 2.21, 2.22, 2.23-rc4
 - Power adapter being used: Amazon Basics 65W GaN PD adapter, Lenovo 65W PD laptop adapter, Anker A2054L11 QC3.0 charger

I'm happy to provide additional info - just tell me what you need and how to provide it.

Ralim, 2025-05-19:
Ahhh this is indeed weird, and definitely not expected.

Can I ask in the debug menu what tip resistance does the tip get detected as (both good and bad ones if possible)?

With the bad tip installed, can you dump out the raw tip reading number from the debug menu as well

c2h5oh, 2025-05-19:
Bad Tip (TS-B2 model - pinecil v2 default):
- Tip Res: 0.0
- Tip R: fluctuating around 3000 +/-100

Good tip (TS-BC2 model):
- Tip Res: 6.2
- Tip R: fluctuating around 650 +/-50

Ralim, 2025-05-27:
That indicates that the TS-B2 tip is likely open circuit; any chance you have a multimeter or means of measuring the tip resistance?