Ralim/IronOS #2146: TS101 PD negotiation failure

0Karakurt0, 2025-07-25:
**Describe the bug**

TS101 fails to negotiate powering on Baseus GaN2 Lite Quick Charger. It boots, but it only sees 5 volt and is interrupted after couple of seconds. None of PD modes from settings worked.
Official firmware does not fail to negotiate Power Delivery and works correctly.

**To Reproduce**

It would negotiate PD and pull all 65 watts available

**Details of your device:**
 
 - Device: TS101
 - Release: v2.23d.009AA10A
 - Power adapter being used: Baseus GaN2 Lite Quick Charger

**Additional context**

Official manual (yep 😄) notes that it supports "QC4, PD3.0, QC3.0, QC2.0, etc".
Listed power profiles are:
- 5v 3A
- 9v 3A
- 12v 3A
- 15v 3A
- 20v 3,25A

0Karakurt0, 2025-07-25:
#2138
#2109
#2104
#2069
#2041
#1887
#1851
#1668
#1680

Considering all of those issues, I suspect that there is something wrong with our PD implementation.
What is needed to debut that?

0Karakurt0, 2025-07-25:
> At this point in time the limiting factors are the ability to debug said chargers (For ultimate debugging I need I2C and PD bus captures). These issues are compounded by my available time, motivation and very very few people contributing to help.

What is needed to capture those?

USB-C interpose, I assume?

Ralim, 2025-07-25:
Two devices are typically required:
1. A USB-PD capture device, that can record messages with timestamps.
2. A logic analyser that can capture the raw I2C messages on the I2C bus. + USB-C breakout that can be inserted inline to tap I2C on Pinecil devices or soldered wire onto I2C pullups on other devices to gain access to the internal I2C bus.

I can guess debug chargers with (1), but (2) is generally required if its not a simpler issue. Note: Only Pinecil devices have the I2C bus exposed for easy debugging.

laro-dev, 2025-08-17:
I tried v2.23D with succes. It now starts directly. However it does not use more than 15V. Previously it was 20V. Maybe this is a setting I am missing.
Anyway, the main problem for me is solved. Thank you.

Ralim, 2026-08-18:
You can grab the assets from the Github CI's latest run for `dev` [here](github.com/Ralim/IronOS/actions/runs/30195635954)

EyalRotem, 2026-08-25:
As per my last comment- the latest dev still resetsthe iron, yet release 2.22 works (same charger, ts101)

abjugard, 2026-08-25:
Yeah I think the consensus is the PD stack in general is the problem with all of these PD-related requests. Someone will probably need to spend a bunch of time replacing the implementation if we want to solve them.

I'm just gonna keep running OFW on my TS101 sadly, and for my Pinecil V2s I'll just have to find chargers that work for them.

Ralim, 2026-08-25:
For all of the PD issues I've seen so far, its been the iron failing to meet the PD turn-around time before the next message. 

I do not have a TS101 to test or capture the I2C traffic on to debug unfortunately. I would love it if someone would be willing to help diagnose these but I'm currently unable to do so due to both not having a device that shows the issue on hand and also not being anywhere remotely near my normal workbench for the time being to work on these things.

A PD capture would help confirm what is going on; and then can try and figure out avenues for debugging further

abjugard, 2026-08-25:
I could try to figure out how to capture a full PD handshake for the TS101 running OFW and IronOS, I think that's what my [FNIRSI FNB58](https://www.fnirsi.com/products/fnb58) is supposed to do haha.

EyalRotem, 2026-08-25:
> For all of the PD issues I've seen so far, its been the iron failing to meet the PD turn-around time before the next message.
>
> I do not have a TS101 to test or capture the I2C traffic on to debug [...]

I can try and help, but unfamiliar on how to capture said data. If special hardware is needed, then no luck, if SW and pc, or esp devices are able to help, I can try

0Karakurt0, 2026-08-25:
@EyalRotem please read https://github.com/Ralim/IronOS/issues/2146#issuecomment-3117538752

0Karakurt0, 2026-11-14:
I have the good news and the bad news)
The good news is that I finally got time to get Power-Z logger working (on windows -_-) and now I can see Power Delivery negotiations.

The bad news is that I no longer have a problematic power supply)
The worst I have is a powerbank of mine that cuts power after a minute of no load, but stock firmware is also affected, despite chattering continiously.

Considering that most of the issue is hadled, I think it is time to rewrite the usefull comments as a wiki entrly on PD debugging and close it.